Xinyuan-LilyGO/T-Embed #5: Unable to find correct settings for TFT_eSPI for this display

Gun-neR, 2022-12-08:
I have loaded the files into my Library as per this pages instructions (I am using a portable IDE so as to not mess up my other displays that use same library), but the User_Setup.h file appears to be incorrect for this display's SPI interface.  I have tried many adjustments but still unable to get anything to display.  At most I have only gotten the backlight to turn on.

What is the correct User_Setup.h for this Embed display?

mmMicky, 2022-12-08:
https://github.com/Xinyuan-LilyGO/T-Embed/blob/main/lib/TFT_eSPI/User_Setups/Setup24_ST7789.h

I used this configuration. I did not create a new Setupxxx.h.

Gun-neR, 2022-12-08:
I did see those options, but have no idea how to chose that file instead.  Even takeing those port numbers and using those in the User_Setup.h doesn't seem to work.   Frusteratingm, as I have been able to get the T-Display runing just fine via that file.

Gun-neR, 2022-12-08:
This is what I have in my User_Setup.h   No idea what it doesnt seem to work.

```
//                            USER DEFINED SETTINGS

#define USER_SETUP_INFO "User_Setup"
#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display
//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For ST7789, ST7735, ILI9163 and GC9A01 ONLY, define the pixel width and height in portrait orientation
#define TFT_WIDTH  170 // ST7789 172 x 320
#define TFT_HEIGHT 320 // ST7789 240 x 320

// For ESP32 Dev board (only tested with GC9A01 display)
// The hardware SPI can be mapped to any pins
#define TFT_MOSI 11 // In some display driver board, it might be written as "SDA" and so on.
#define TFT_SCLK 12
#define TFT_CS   10  // Chip select control pin
#define TFT_DC   13  // Data Command control pin
#define TFT_RST  9  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   16  // LED back-light

// ##################################################################################
//
// Section 3. Define the fonts that are to be used here
//
// ##################################################################################
// Comment out the #defines below with // to stop that font being loaded
// The ESP8366 and ESP32 have plenty of memory so commenting out fonts is not
// normally necessary. If all fonts are loaded the extra FLASH space required is
// about 17Kbytes. To save FLASH space only enable the fonts you need!
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

// ##################################################################################
//
// Section 4. Other options
//
// ##################################################################################
// Define the SPI clock frequency, this affects the graphics rendering speed. Too
// fast and the TFT driver will not keep up and display corruption appears.
// With an ILI9341 display 40MHz works OK, 80MHz sometimes fails
// With a ST7735 display more than 27MHz may not work (spurious pixels and lines)
// With an ILI9163 display 27 MHz works OK.
#define SPI_FREQUENCY  20000000
#define SPI_READ_FREQUENCY  2000000
```

mmMicky, 2022-12-09:
The role of this file is to define your lcd pin definitions. You can modify it. Modify it to match my definition.

Gun-neR, 2022-12-09:
I know how the file works :)  I just don't like its complexity.  And even after trying all sorts of GPIO combinations and SPI frequencys, the best I could do with the display was enable/disable the back-light ;P

Even using an Adafruit display driver didn't work... although with that it at least showed a brief flash of scrambled "screen garbage" every once in awhile.

If I hadn't at lest seen the initial demo I would be thinking that it is a faulty device (and it still might be??)  Anyhow, I think this latest acquisition of mine from LILYGO is LILYbound for the bin or a dark corner of a drawer until there is more matured info/examples to test with, perhaps using Python or something, and/or I feel like being frustrated for another 2+ days straight by a silly MCU :D

mmMicky, 2022-12-09:
If you don't like modification. I have provided already modified TFT_eSPI library file, why not use it.
I use the platformIO tool in the build factory. I recommend using it.

Gun-neR, 2022-12-09:
As I already inferred, I tried your data, my data, others data... still didn't work, nor did a totally different library.  And I don't like or use PlatformIO :)

As I only ordered the single one of this T-Embed, I can only assume it is either a faulty device or I am missing some small un/poorly documented detail...  Ie. The instructions and settings that come with this library infer 8bit parallel but document claims SPI... tried both.

I have tested all other aspects of the device, except the mic.  So far only the display is not working, which basically renders the whole thing as useless for the time being.   I will set it aside for now.

closed 2022-12-09 by Gun-neR

Gun-neR, 2022-12-10:
OK, thank you for the pin-out info, if not for the lack of clarification that your version of the Setup24_ST7789.h file WASN'T the same as the "official" one in the actual library (Hint - Clarity means everything in a good answer ;P )  But once I realized the discrepancy and edited the original Setup24_ST7789.h file to match your "custom" version of it, everything worked.

Strange that it wouldn't work with those same pin-outs and freq in my custom User_Setup.h file??  Probably something else in that convoluted library was messing up my attempts.  

And no idea why Adafruits example only showed glitching... I suspect there is something different in the way the T-Embed hooks up to that LCD, and my figuring out that one is for another day and not relevant for this topic.

Anyhow. now it is working and so back to my regularly shedualed programming :D

mattoppenheim, 2024-01-04:
My Lilygo T-Embed-S3 remained blank until I added the lines in bold to the TFT_eSPI examples supplied by Lilygo. I build and load these files using the platformio plug-in with VSCode running on Debian.

>**#define PIN_POWER_ON          46**
void setup(void) {
 >>**pinMode(PIN_POWER_ON, OUTPUT);
      digitalWrite(PIN_POWER_ON, HIGH);**
      tft.init();