Xinyuan-LilyGO/T-Embed #7: PWR_ON pin function is not documented

cydergoth, 2023-01-15:
It turns out to be quite critical to know about when trying to write your own application from scratch!

```
#include "driver/gpio.h"

#define PIN_POWER_ON          46

void app_main(void)
{
    gpio_config_t pw_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << PIN_POWER_ON
    };
    ESP_ERROR_CHECK(gpio_config(&pw_gpio_config));
    gpio_set_level(PIN_POWER_ON, 1);  // <--- LCD won't start without this!

 ....
}
```

cydergoth, 2023-01-15:
Fully working demo with ESP-IDF V5.0 and LVGL 8.3.0 (add it as a dependency with `idf.py add-dependency lvgl/lvgl`)

Uses the `img_logo.h` from the Arduino demo

```
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "hal/spi_types.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_commands.h"
#include "driver/gpio.h"
#include "driver/spi_master.h"
#include "esp_err.h"
#include "esp_log.h"
#include  "lvgl.h"
#include "esp_timer.h"
#include "img_logo.h"

#define PIN_POWER_ON          46

// ST7789 Parameters not documented in ESP IDF esp_lcd_panel_commands
#define ST7789_RAMCTRL		0xB0      // RAM control
#define ST7789_PORCTRL		0xB2      // Porch control
#define ST7789_GCTRL		0xB7      // Gate control
#define ST7789_VCOMS		0xBB      // VCOMS setting
#define ST7789_LCMCTRL		0xC0      // LCM control
#define ST7789_IDSET		0xC1      // ID setting
#define ST7789_VDVVRHEN		0xC2      // VDV and VRH command enable
#define ST7789_VRHS		0xC3      // VRH set
#define ST7789_VDVSET		0xC4      // VDV setting
#define ST7789_VCMOFSET		0xC5      // VCOMS offset set
#define ST7789_FRCTR2		0xC6      // FR Control 2
#define ST7789_PWCTRL1		0xD0      // Power control 1
#define ST7789_VAPVANEN		0xD2      // Enable VAP/VAN signal output
#define ST7789_CMD2EN		0xDF      // Command 2 enable
#define ST7789_PVGAMCTRL	0xE0      // Positive voltage gamma control
#define ST7789_NVGAMCTRL	0xE1      // Negative voltage gamma control
#define ST7789_DGMLUTR		0xE2      // Digital gamma look-up table for red
#define ST7789_DGMLUTB		0xE3      // Digital gamma look-up table for blue
#define ST7789_GATECTRL		0xE4      // Gate control
#define ST7789_SPI2EN		0xE7      // SPI2 enable
#define ST7789_PWCTRL2		0xE8      // Power control 2
#define ST7789_EQCTRL		0xE9      // Equalize time control
#define ST7789_PROMCTRL		0xEC      // Program control
#define ST7789_PROMEN		0xFA      // Program mode enable
#define ST7789_NVMSET		0xFC      // NVM setting
#define ST7789_PROMACT		0xFE      // Program action

// Why is this called LCD PIXEL CLOCK when it's really SPI Bus speed?
#define LCD_PIXEL_CLOCK_HZ     (20 * 1000 * 1000)

// LVGL update interval
#define LVGL_TICK_PERIOD_MS 2

#define TAG "hello_lcd"

// from https://github.com/Xinyuan-LilyGO/T-Embed
#define LCD_H_RES              170
#define LCD_V_RES              320

// Commands for the LCD panel on init
typedef struct {
    uint8_t cmd;
    uint8_t data[14]; // TODO: Seems like a waste of space, can do better surely
[... 225 lines cut ...]
    ESP_LOGI(TAG, "Install LVGL tick timer");
    // Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
    const esp_timer_create_args_t lvgl_tick_timer_args = {
        .callback = &increase_lvgl_tick,
        .name = "lvgl_tick"
    };
    esp_timer_handle_t lvgl_tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, LVGL_TICK_PERIOD_MS * 1000));

    // Ensure the coordiate systems align with the physical display
    lv_disp_set_rotation(disp, LV_DISP_ROT_270);

    ESP_LOGI(TAG, "Display LVGL");
    lvgl_demo_ui(disp);

    while (1) {
        // raise the task priority of LVGL and/or reduce the handler period can improve the performance
        vTaskDelay(pdMS_TO_TICKS(10));
        // The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
        lv_timer_handler();
    }
}

```

cydergoth, 2023-01-15:
Also can't find any documentation on LCD command 0xD6

cydergoth, 2023-01-17:
Uploaded the above demo as a proper PR, having improved the structure to make it more reusable

lewisxhe, 2023-03-17:
"Yes, it depends on the hardware. The prerequisite for using it is to set IO46 to HIGH. Thank you for your example. If you can submit a PR for me, I will close the issue."

closed 2023-03-17 by lewisxhe