Xinyuan-LilyGO/T-Embed #16: Is there a 'boot' button on the T-Embed board as I need to reset

reflectingme, 2023-02-28:
On the t-display boards there is a boot button that can be used to reset the board in the event of a fail, by holding it down before plugging the USB cable in, then uploading the code etc.

How does one do this on the T-Embed board please?

Thanks

Maestro8484, 2023-03-03:
i would like to know this as well.  how does one activate this 'boot' function AND how do you turn the device off when on battery power?

reflectingme, 2023-03-07:
So, I have discovered the BOOT button GPIO 0 is actually the centre button for the rotary encoder dial.  However, using the same process as described on the T-Display GitHub board page to reset the board via BOOT mode does not work with the T-Embed board.
Frustrating.

closed 2023-03-15 by reflectingme

lesept777, 2023-09-01:
It seems possible to reset the T-Embed. GPIO 0 is connected to the encoder button.
Push the encoder button, push the reset button, power on the device, release the reset button then release the encoder button