Xinyuan-LilyGO/T-Embed #22: Grove connector pinout on reference diagram does not match board

cydergoth, 2023-04-06:
On the reference diagram the Grove connector has GND 3V3 IO44 IO43, however on the physical board the pinout legend reads GND IO44 IO43 3V3.

 From the Grove documentation it seems as if the physical board has the correct pinout.

lewisxhe, 2023-04-09:
I will conduct an investigation on Monday

lewisxhe, 2023-04-10:
Please connect according to the silk screen instructions of the physical board. The pin diagram is updated later on the board.

closed 2023-04-10 by lewisxhe

akudan, 2023-04-30:
This is particularly confusing because the silkscreens are different between the image on GitHub and on lilygo.cc/Tindie. If the latter silkscreens are correct (i.e. GND IO44 IO43 3V3) then it's actually not directly Grove compatible and you'd need to modify a cable.