Xinyuan-LilyGO/T-Embed #23: Q: OTG host with power

cydergoth, 2023-04-08:
I'm trying to use the T-Embed with a USB device, using OTG.

In order to power it I have a USB OTG/PD cable which has a Y connector breaking out a USB-C port for power and a USB-A port for OTG.

Unfortunately when using this cable the T-Embed will not power up. Is there a suggested way to make this configuration work? I don't think I can use a battery as the battery voltage is 3.7V and the USB device needs 5v. 

The cable is similar to this one: https://www.amazon.com/Adapter-Charging-Compatible-XL%EF%BC%8CGoogle-Chromecast/dp/B08C5FWQND

lewisxhe, 2023-04-09:
This may be because the cable is not supplying power to the t-embedded. You can use a multimeter to measure the position of the 5V Pin on the board and check if there is 5V

lewisxhe, 2023-04-13:
[image]
The easiest way is to measure the USBC here, be careful not to short circuit

cydergoth, 2023-04-15:
I tested the point you indicated and it is not receiving 5v. It looks like the PD negotiation is not working. Unfortunately I am not an expert at USB-C so I'm not sure if this is possible to fix in the bootloader or if it requires additional hardware to perform the negotiation

cydergoth, 2023-04-15:
If I plug in the upstream cable without the splitter, it powers up ok

cydergoth, 2023-04-15:
From my limited understanding of USB-C and looking at the schematic, it looks to me as if the T-Embed correctly implements the pulldown resistors for the "USB Type-C UFP sink: USB 2.0 without USB PD". This, along with it working with the upstream cable, implies to me that it is the splitter cable which is incorrect, so I'm closing this issue. Thanks for the useful feedback!

closed 2023-04-15 by cydergoth