Xinyuan-LilyGO/T-Embed #26: How to reset?

d3mac123, 2023-06-30:
I ran my first code (sample from here) into the board but nothing shows on the display (only the blue led blinks). I then tried the audio sample and all I get, besides the black screen, is a noise (like a dot) from the speaker :(

lewisxhe, 2023-06-30:
Check whether TFT_eSPI/User_Setup_Select.h has been correctly configured embed https://github.com/Xinyuan-LilyGO/T-Embed/blob/50faf3ed2911faf423489a9e3045e45eb219f808/lib/TFT_eSPI/User_Setup_Select.h#L139

closed 2023-07-04 by d3mac123

lesept777, 2023-09-01:
It seems possible to reset the T-Embed. GPIO 0 is connected to the encoder button.
Push the button, push the reset button, power on the device, release the reset button then release the encoder button