Xinyuan-LilyGO/T-Embed #27: How to run the music example?

d3mac123, 2023-07-04:
I am trying to run the music (play mp3 file) example but I am getting the following error (after installed the ESP32-audioI2S library). Any idea on how to fix this? I am using Arduino IDE 2.1.
`/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x30): undefined reference to `MP3Decoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x38): undefined reference to `AACDecoder_IsInit()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x3c): undefined reference to `AACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x40): undefined reference to `FLACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x44): undefined reference to `OPUSDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio17initializeDecoderEv+0x48): undefined reference to `VORBISDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x28): undefined reference to `AACGetFormat()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x2c): undefined reference to `AACGetID()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x30): undefined reference to `AACGetProfile()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x18): undefined reference to `MP3FindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x1c): undefined reference to `AACFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x20): undefined reference to `AACSetRawBlockParams(int, int, int, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x24): undefined reference to `FLACSetRawBlockParams(unsigned char, unsigned int, unsigned char, unsigned int, unsigned int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x28): undefined reference to `FLACFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x2c): undefined reference to `OPUSFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x30): undefined reference to `VORBISFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x1c): undefined reference to `MP3GetBitrate()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x20): undefined reference to `AACGetBitrate()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/A467C159EA95225C167DF9EE4CCB4D3C/libraries/ESP32-audioI2S-master/Audio.cpp.o:(.literal._ZN5Audio24compute_audioCurrentTimeEi+0x24): undefined reference to `FLACGetBitRate()'
[... cut ...]
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/alexsouza/Library/Arduino15/libraries/ESP32-audioI2S-master/Audio.cpp:327: undefined reference to `VORBISDecoder_FreeBuffers()'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
  Used: /Users/alexsouza/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD
  Not used: /Users/alexsouza/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1`

lewisxhe, 2023-07-05:
[image]
[image]

I didn't encounter your situation, please make sure you are using the version library that matches mine
https://github.com/Xinyuan-LilyGO/T-Embed/tree/main/lib/ESP32-audioI2S

d3mac123, 2023-07-06:
Is there a proper way to install this library? I manually created all the folders and files inside my Mac's Library/Arduino15/libraries/ESP32-audioI2S, set the Arduino IDE as you shared above, and ran the sample code again. This is what I am getting now:
`/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15read_OGG_HeaderEPhj+0x1c): undefined reference to `FLACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio16parseContentTypeEPKc+0x48): undefined reference to `MP3Decoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio16parseContentTypeEPKc+0x4c): undefined reference to `AACDecoder_IsInit()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio16parseContentTypeEPKc+0x50): undefined reference to `AACDecoder_AllocateBuffers()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x2c): undefined reference to `AACGetFormat()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x30): undefined reference to `AACGetID()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio15showCodecParamsEv+0x34): undefined reference to `AACGetProfile()'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x18): undefined reference to `MP3FindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x1c): undefined reference to `AACFindSyncWord(unsigned char*, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x20): undefined reference to `AACSetRawBlockParams(int, int, int, int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x24): undefined reference to `FLACSetRawBlockParams(unsigned char, unsigned int, unsigned char, unsigned int, unsigned int)'
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /private/var/folders/j0/_j5h8rw161l2lrx2khy0_2wh0000gn/T/arduino/sketches/77341F499301C74B78064687114E8769/libraries/ESP32-audioI2S/Audio.cpp.o:(.literal._ZN5Audio12findNextSyncEPhj+0x28): undefined reference to `FLACFindSyncWord(unsigned char*, int)'
[... cut ...]
/Users/alexsouza/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld: /Users/alexsouza/Library/Arduino15/libraries/ESP32-audioI2S/Audio.cpp:733: undefined reference to `FLACDecoder_AllocateBuffers()'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
  Used: /Users/alexsouza/Library/Arduino15/packages/esp32/hardware/esp32/2.0.9/libraries/SD
  Not used: /Users/alexsouza/Library/Arduino15/libraries/SD
exit status 1

Compilation error: exit status 1`

closed 2023-07-29 by d3mac123