Xinyuan-LilyGO/T-Embed #28: Power can't be turned off

FrantisekBeranek, 2023-07-26:
- You can't turn power off because Q2 is not connected properly (Drain and Source are swapped). It means that V_BAT is still on thru internal diode of this transistor when you set pin PWR_ON to 0. 
- The resistor RQ is not wired properly. It should be connected to BAT not V_BAT so it still keep Gate of Q2 high when power is off.
- Voltage divider for BAT_ADC should not (in my opinion) be connected to BAT directly because if you would be able to turn MCU down you can possibly destroy your ADC input by applying voltage in it without power. Also it drains your battery constantly (current is small but it is there).
- There is no other option to power up the board than to connect USB (you apply VBUS on Q3 which starts U4). Of corse it's not problem when you can't turn the board off but I think button in parallel with Q5 would be nice.
- Applying VBUS to Gate of Q3 actualy closes the transistor so U4 is powered from VBUS and then from V_BAT as mentioned above.
- It seems to me that VBAT and BAT should be the same net.

I can be wrong in any of these, so please don't be angry for useless issue.

lewisxhe, 2023-07-27:
Thanks for your feedback, I will give feedback to the hardware engineer responsible for this project.

lilysam23, 2023-08-13:
T-embed esp32 s3 bilgisayara taktım ve açılmımamaya başladı

closed 2023-10-18 by lewisxhe

radimsejk, 2023-10-18:
@lewisxhe How is this completed? What is the result please?

lewisxhe, 2023-10-18:
I have given feedback to the hardware and this issue can be closed.