Xinyuan-LilyGO/T-Embed #29: Sound example not playing anything

d3mac123, 2023-07-29:
Hi, I installed everything as per the instructions and I am trying to run the sound example. I do not get any errors in the Output window neither in the Serial Monitor (see what the serial monitor shows below) but, no sound comes from the T-embed speakers. Any ideas what may be causing this?

Serial monitor:
`psram size : 8189 kb
FLASH size : 16384 kb`

orion14794, 2023-07-29:
> Hi, I installed everything as per the instructions and I am trying to run the sound example. I do not get any errors in the Output window neither in the Serial Monitor (see what the serial monitor [...]

I get the exact same issues and I've reflashed the sketch several times... Hope they fix the issues with the code/sketch asap.

lewisxhe, 2023-10-18:
Has the file system been uploaded?