Xinyuan-LilyGO/T-Embed #32: Problem with flashing firmware.bin using Espressif tool

pganson, 2023-10-16:
I downloaded the Espressif flash download tool, v3.9.4 and the T-Embed project from this github repository.  I am able to use the "erase" button and also to flash the firmware.bin file to my T-Embed. 

But, then when I reboot the T-Embed, I get "Invalid header: 0xFFFFFFFF" continuously on the serial output from the T-Embed.  The T-Embed appears to be in a reboot loop; the USB device on my computer keeps disconnecting and re-connecting.

lewisxhe, 2023-10-18:
[image]
Are the settings the same?

pganson, 2023-10-18:
Yes, my settings are the same as instructed. 

[image]

lewisxhe, 2023-10-19:
You did not check the firmware you need to burn.
[image]

pganson, 2023-10-30:
I'm getting a timeout during the download of the firmware.bin image to the T-Embed.  I've tried changing the baud rate (all the way to 115200) but it has no effect.  I also tried two different ports on my laptop - USB-A and USB-C.

Do you have any ideas for something else to try ?

[image]

NaegeliJ, 2024-05-12:
Hi @pganson, I encountered the same issue. Espressif released an updated version of their firmware flash tool which resolved the problem for me. You can download it here: [Espressif Flash Download Tool 3.9.6](https://www.espressif.com/sites/default/files/tools/flash_download_tool_3.9.6_1.zip). Hope this helps