Xinyuan-LilyGO/T-Embed #34: GPIO Pinout reference

DMFSouza, 2024-01-26:
I recently acquired a T-Embed, but when trying to add an RC522 module, I came across this doubt: how do I reference the GPIO pins? Which numbering corresponds to the MISO, CLK, and SDA pins? I searched the manual, but the numbering doesn't seem to be correct. Thank you very much for your attention and time in advance!
[image]
[image]

lewisxhe, 2024-01-27:
If you need to use this SPI, then there is a problem. This hardware design does not reserve an extra free CS pin. You need to connect CS to GPIO 44 or 43.
Regarding which MISO MOSI and SCK are SPI, you can see here.
https://github.com/Xinyuan-LilyGO/T-Embed/blob/606694fd07a5bf377caef45fe7fd5685549bc3e1/examples/CC1101_Receive/TFTHelper.h#L46

reviwer-ai, 2025-06-03:
> [image] Is this OK? 16, 17 are free, MISO, MOSI, SCK are SPI bus

I am using bruce firmware on t embedded can you please write the connection digram EBYTE E07-433M20S Ebyte CC1101 Module, SMD Long Range Wireless RF Transceiver Module please full connection

lewisxhe, 2025-06-03:
I can only provide the corresponding locations of these IOs. Please check other modules yourself.

reviwer-ai, 2025-06-03:
> I can only provide the corresponding locations of these IOs. Please check other modules yourself.

Can you tell the pin connections and where it located I dont want to disable sd card slot 
Miso 
Mosi 
Sck
Cs i didnot find it 
Gdo0 irq slots can you mark where it located

lewisxhe, 2025-06-03:
> If you need to use this SPI, then there is a problem. This hardware design does not reserve an extra free CS pin. You need to connect CS to GPIO 44 or 43. Regarding which MISO MOSI and SCK are SPI, you can see here.
> 
> [T-Embed/examples/CC1101_Receive/TFTHelper.h](https://github.com/Xinyuan-LilyGO/T-Embed/blob/606694fd07a5bf377caef45fe7fd5685549bc3e1/examples/CC1101_Receive/TFTHelper.h#L46)
> 
> Line 46 in [606694f](/Xinyuan-LilyGO/T-Embed/commit/606694fd07a5bf377caef45fe7fd5685549bc3e1)
> 
>  // SPI Share with SDCard , NFC , Radio(CC1101) 
> Please note that GPIO17,18 are I2C, if used for other purposes, the onboard decoding ES7210 will not be able to communicate.

I think it is explained here Using GPIO 43,44

reviwer-ai, 2025-06-03:
> I think it is explained here Using GPIO 43,44

Can you tell the spi pins connections 
Without modification the firmware

reviwer-ai, 2025-06-03:
> You need to check the source code of Bruce firmware. I don't know much about Bruce.

Can you tell me the sd card slot connection I am confused sd card have 7 pins witch pin is cs  sclk mosi miso can you give me image diagram

reviwer-ai, 2025-06-03:
> You need to check the source code of Bruce firmware. I don't know much about Bruce.

> > [image] Is this OK? 16, 17 are free, MISO, MOSI, SCK are SPI bus
> 
> I am using bruce firmware on t embedded can you please write the connection digram EBYTE E07-433M20S Ebyte CC1101 Module, SMD Long Range Wireless RF Transceiver Module please full connection

Can you give the sd card slot diagram like this photo I want to see the diagram

reviwer-ai, 2025-06-03:
> [image]

Not like this like the pic you sent earlier

lewisxhe, 2025-06-03:
[image]

reviwer-ai, 2025-06-08:
> [image]

Can you give me the antenna pcb where does the antena come from photo give me diagram like this

reviwer-ai, 2025-06-09:
> There is no antenna on the embed board, only an antenna on the esp module. What do you mean?

[image]
I mean I broke this pin I marked and my t embedded not detecting wifi networks and I want to know where I can attach the external antena and give me your instagram id so I can give you more reason

lewisxhe, 2025-06-09:
This metal solder joint is the antenna outlet. This is the ESP32 module. It is not our board. This is from Espressif. If the module is not damaged, it should work.
Sorry, I don't have Instagram

reviwer-ai, 2025-06-09:
> This metal solder joint is the antenna outlet. This is the ESP32 module. It is not our board. This is from Espressif. If the module is not damaged, it should work. Sorry, I don't have Instagram

[image]
[image]

This is my t embedded and this is the condition of the antena I am saying how to attach antenna on this the pcb is damaged and the solder is gone cause heat

lewisxhe, 2025-06-10:
[image]

Oh my goodness.
Maybe you can try soldering the antenna to this capacitor. Solder the shield to GND
If that doesn't work then it's probably damaged.