Xinyuan-LilyGO/T-Embed #38: Readme for Arduino Quick Start Doesn't Work for Newer IDE

mrmonteith, 2024-07-16:
On the T-Embed - Readme under Quick Start - Arduino the instructions do not work on the new versions of the IDE.  
[T-Embed](https://github.com/Xinyuan-LilyGO/T-Embed)
 It seems to go through the steps to install.  The search fails
Search for ESP32 in Tools->Board Manager and install ESP32-Arduino SDK 

 Is this the one that is needed now?
[image]

LilyGO, 2024-07-17:
Yeah, that's it.

closed 2024-07-19 by mrmonteith

mrmonteith, 2024-07-19:
Thanks.  I supposed the next line "Copy all files in the lib folder to \Arduino\libraries" doesn't apply?  Seeing there is no lib folder.  Needs to explain where lib is.  Thanks