Xinyuan-LilyGO/T-Embed #45: New Display for T-Embedded Compatibility Issue

David-Feldt, 2025-04-01:
I have a T-Embeded V1.2 and ordered a new LCD screen (I broke the old one). I plugged in the ribbon cable and ran the default example and the display is not working. The backlight is working but no actual data. Any help is appreciated.

David-Feldt, 2025-04-01:
I purchased from lilygo, is there any diagnostics I can run?

lewisxhe, 2025-04-02:
T-Embed has two versions, one is T-Emed and the other is T-Embed-CC1101. First, you need to confirm that your version is the current version of this repository.

Then you can directly write [binary firmware](https://github.com/Xinyuan-LilyGO/T-Embed/blob/main/firmware/firmware.bin) to determine whether the board can run normally

https://github.com/Xinyuan-LilyGO/T-Embed/tree/main/firmware

David-Feldt, 2025-04-04:
This worked thanks so much!