Xinyuan-LilyGO/T-Embed #54: cc1101 for t embed

igniskarya, 2025-07-02:
hi there...

I want connect cc1101 on my t embed. the cc1101 module like this

[image]

and if i looked at on example  

#define SHIELD_RADIO_CS_PIN                 17
#define SHIELD_RADIO_DIO0_PIN               18
#define SHIELD_RADIO_DIO2_PIN               8
#define SHIELD_RADIO_SW1_PIN                41
#define SHIELD_RADIO_SW0_PIN                43
#define SHIELD_RADIO_RST_PIN                -1

is it possible to use that cc1101 module ?
but I little bit confuse which pin should I connect for
pin GD00, SCk,MOSi,MiSO, GD02

could you give me idea about those pins

igniskarya, 2025-07-04:
> hi there...
>
> I want connect cc1101 on my t embed. the cc1101 module like this
>
> [image]
>
> and if i looked at on example
>
> #define SHIELD_RADIO_CS_PIN 17 #define SHIELD_RADIO_DIO0_PIN 18 #define [...]

if looked at on that pin ...
DiO2 pin 8.. 
I try looking for that pin...in which part pin 8 position?

lewisxhe, 2025-07-08:
Well, no more GPIO, this board wastes a lot of GPIO